RomanLut/INAV-X-Plane-HITL #8: [Task] build plugin for MacOs

RomanLut, 2022-07-05:
Build plugin for MacOs

RomanLut, 2022-07-08:
Hint: start from project: https://developer.x-plane.com/code-sample/openal-shared-example/
use #if APL srections for MasOs specific code.

krl91, 2022-08-08:
Hi, I try to build a version on MacOS (last version with a Mac-Intel) but i'm stuck:
I installed openal with brew, maybe not the best idea
my modifications:
[diff.txt](https://github.com/RomanLut/INAV-X-Plane-HITL/files/9284919/diff.txt)

The error I have:
`ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [build/INAV-X-Plane-HITL/64/lin.xpl] Error 1`

bartslinger, 2024-07-08:
For what it's worth, I managed to compile something. It's very raw, just trying to get something to work before optimizing. However X-Plane 11 just crashes when starting a flight. Might look further into this later.

https://github.com/bartslinger/INAV-X-Plane-HITL/commit/91970246678384fa629f6aeccb3e496c3e1b8fa5

RomanLut, 2024-08-13:
Merged into master

closed 2024-08-13 by RomanLut