RomanLut/INAV-X-Plane-HITL #9: Can't connect to Inav-Configurator on Linux after connection with HIL

sevrugin, 2022-07-23:
Steps to reproduce:
1 run XPlane and connect to flight-controller (from plugin menu)
2 disconnect plugin (from plugin menu)
3 try to connect with inav-confaigurator - failed

But! If after the step 2 we will connect to the port by terminal:
`screen /dev/ttyACM0 115200` and then `killall screen`
we will be able to connect by inav-configurator

In my mind, after we closed the serial port in plugin we still have some unflashed data 
Also, it would be great to add `if (isConnected())` to each methods

lwan288, 2023-04-02:
I have HITL installed under x-plane 12 on Ubuntu 22.04. However, HITL doesn't show up under plugin options. I have tried the same on Windows 10. HITL works perfectly.    Not sure which step I did wrong on Linux system. Or, is there anyone having the similar issue?

[image]
[image]
[image]

lwan288, 2023-04-02:
> 

I used chmod for FC connection with inav-configurator on Ubuntu and it works well, but HITL doesn't work on my Ubuntu machine. Not sure what happened. It looks there is a compatibility issue with lin.xpl on my Linux machine. 

sudo  chmod 0777 /dev/ttyACM0

RomanLut, 2023-04-05:
(I am not using Linux and I am not an exprt in Linux)
One guess is that some libraries are missing. 

Try running:
ldd lin.xpl

lwan288, 2023-04-05:
> (I am not using Linux and I am not an exprt in Linux) One guess is that some libraries are missing.
> 
> Try running: ldd lin.xpl

Yes, libopenal.so.1 is missing. I installed the library using the following command and it works!  
sudo apt install libopenal1

Thanks so much!!