RomanLut/INAV-X-Plane-HITL #11: X-Plane Attitude causing variometer false readings and impossible althold with new AHRS

b14ckyy, 2022-09-18:
I currently see a issue with the vertical speed indicator when X-Plane direct attitude is enabled. 
Since I test the new AHRS for 6.0, everytime the plane makes a turn, the vario shows extremely high climb rates, rendering position hold useless as it will constantly push the nose down to stabilize the climb instead of holding altitude. Here is a quick video https://www.youtube.com/watch?v=EZ6_PyfUlu8

Full simulation with sensor estimation on the other hand is perfectly fine. The new AHRS seems to handle inconsistent sensor rates very well. 

I am not sure if this is a plugin bug as I recently updated to 1.2 or INAV side bug or only a bug in combination with the new AHRS so I thought I post it here.

RomanLut, 2022-09-18:
Can not reproduce with Inav firmware built from master branch. Is this new AHRS system merged into master?

b14ckyy, 2022-09-18:
> Can not reproduce with Inav firmware built from master branch. Is this new AHRS system merged into master?

No it is not in master yet as Julio is doing some refinements still. But he updated his build with most current PRs from master. I can send you a build for most matek boards.

RomanLut, 2022-09-18:
I do not have matek boards, I have to build for my FC.
Is there any PR for new AHRS?

b14ckyy, 2022-09-18:
Not yet. Not before we know it fully works as we want it to :D

RomanLut, 2022-09-19:
> this should be pretty recent https://github.com/JulioCesarMatias/inav

https://github.com/JulioCesarMatias/inav:master branch is the same as iNavFlight/inav:master. No changes.
Please provide a link to branch which I could use to build firmware for reproduction.

b14ckyy, 2022-09-19:
OK then he does not put the AHRS updates in there yet as well. 
Let's get this AHRS finished first and then check if it causes the trouble when the PR is open. Otherwise Julio would need to look into it himself.

b14ckyy, 2022-09-23:
Oh I nearly forgot. The PR is up now here: https://github.com/iNavFlight/inav/pull/8407

I have a feeling that this issue with the inverted vertical speed when using X-Plane attitude, has something to do with the not working vertical accelleration compensation on the new AHRS. 
Julio has troed to fix the vertical accelleration by inverting the Z-Axis for the accelleration compensation and this did in fact completely solve the AHI drift after strong Z-Axis g-forces. But only for a short time after launch. Because for some reason it breaks the heading estimation and the heading starts to wander around and does not correct itself anymore.

But I will re-use this inverted Z-Acc build he sent me and see if X-Plane attitude works with that one, to confirm this. 

If I'm right, then fixing this accelleration compensation properly will fix the Simulator vspeed issue, AHI drift issue and heading issues alltogether.

RomanLut, 2022-10-03:
Was issue resolved, or I should look at it?

b14ckyy, 2022-10-03:
It was still there in his last commit but as his PR is on hold for now it is not that important at the moment I guess as it can't be merged as it is right now. 
I will check again with Shota's AHRS update build to see if that works with Direct Attitude estimation.

closed 2022-11-15 by b14ckyy