RomanLut/INAV-X-Plane-HITL #16: GPS Failure

tonyyng, 2023-07-27:
Hi.

I'm trying to get this working with a NOXe F4 flight controller.  I've gone through all the steps and I'm getting a "GPS Failure" system message from iNAV:
[image]

I've got a port assigned for a GPS:
[image]

I've tried changing the GPS baud rate and using SoftSerial1 for GPS.  

I also tried overriding the arming error (yaw right) when arming, but that didn't work either.

I do have this working with another flight controller (SpeedyBee F405 Wing), but I wanted to get the NOXe working.

Any other suggestions?  Is the SoftSerial1 port a problem?

Thanks

tonyyng, 2023-07-27:
The NOX FC is a F411, with two hardware uarts.  The SpeedyBee FC is F405, with 6 hardware uarts.  Could the problem be caused by this?

RomanLut, 2023-07-27:
Just checked Inav compiled from tag 6.0.0 against current release of hitl 1.3.0 - works fine for me. I suggest to check with inav tag 6.0.0.

GPS baudrate does not matter. No physical GPS sensor is required.

RomanLut, 2023-07-27:
> The NOX FC is a F411, with two hardware uarts. The SpeedyBee FC is F405, with 6 hardware uarts. Could the problem be caused by this?

Software serial is not a problem. I am using Noxe v1 with two software UARTs without any problem:
[image]

tonyyng, 2023-07-28:
I tried the iNav 6.0.0 Noxe (Target NOX), downloading the firmware from the online repository.  I get the same error.

I also have a Noxe V1 (Target FLYWOOF411, right?)  I hooked it up and updated the FW to iNav 6.0.0.  I get the same error.

Reconnecting the Speedybee F405 Wing, it works fine.  Weird.  I'll just stick with the Speedybee FC for now.

tonyyng, 2023-07-30:
I figured out what the problem was.  You have to calibrate the accelerometers on the FC.  If you don't, the error that appears is "GPS Failure".  I stumbled across the real problem by accidentally removing the GPS sensor on the UART port.  

I toggled MSP Data ON for the UART that has the GPS sensor and clicked "Save and Reboot"
[image]

When you do that, the sensor for that UART becomes "Disabled".
[image]

Then, when iNav started in that configuration, I saw a different system error message: "ACCELEROMETER NOT CALIBRATED".

Going back into Configurator and calibrating solved the problem.  (Remember to put the GPS device back on a port).

I confirmed that was the cause by clearing the calibration settings and trying to start the iNav HITL plugin again.  Same error appears. "GPS Failure"

# reset calibration
set acczero_x = 0
set acczero_y = 0
set acczero_z = 0
set accgain_x = 4096
set accgain_y = 4096
set accgain_z = 4096

closed 2023-07-30 by tonyyng