RomanLut/INAV-X-Plane-HITL #23: Unable to arm: Waiting for GPS fix

xznhj8129, 2024-04-28:
In HITL with real FC and SITL, there will be randomly a "cannot arm, waiting for GPS fix" bug even if everything shows full 3D lock. Restarting everything will usually make it go away, but in SITL this is a complete roadblock and it will just not go away.

RomanLut, 2024-04-28:
1) Are you sure Menu option is not set to "No GPS Fix" ?

2) Which SITL executabe are you using?

xznhj8129, 2024-04-28:
1. Yes that has been checked, it is always on 3D Fix.
2. SITL started from Inav configurator
I noted that setting GPS to Ublox instead of FAKE will *partially* solve the issue, it will appear ~25% of the time. Issue still appears with HITL Speedybee F405 Wing Mini after a new flight and FC reset, Xplane has to be restarted to make it go away.

lwan288, 2025-01-09:
I have a similar issue. : (  
[image]

I am using the inav 7.1.2.   However, when I switched back to inav 6, the error disappeared.

Hermi-hub, 2025-02-03:
https://github.com/iNavFlight/inav/issues/10086