RomanLut/INAV-X-Plane-HITL #25: No thrust on pitch up

Yury-MonZon, 2024-05-08:
Hi Roman! Thanks for the great possibility to have HITL. It is amazing.

I'm using xplane 11 (getting 60-75fps), Inav 7.1.1 and the wing packaged in release 1.4.0 - there is a issue. 

When I pitch the plane up motor sound shuts off and even with the full throttle (I can see it on OSD) I get no thrust. This stands the same for manual, cruise, acro and angle modes.

RomanLut, 2024-05-09:
Works fine for me with 7.1.1. Please provide "diff all" dump.

Yury-MonZon, 2024-05-09:
[diff_all.txt](https://github.com/RomanLut/INAV-X-Plane-HITL/files/15264872/diff_all.txt)
Here is my diff. I've tried tuning PIDs here. 

But I also tried your PIDs from https://github.com/RomanLut/INAV-X-Plane-HITL/blob/main/doc/nkfpvsufrwing/nkfpvsurfwing.md with the same results.

Yury-MonZon, 2024-05-09:
Found the issue - I wasn't starting the engines, my hotkey macro was broken. 

**For those who will face the same issue - press Shift+Control+e**

@MrD-RC I've tried that noob wiing too - for me it oscillates even in manual mode. Can you provide your settings please?

MrD-RC, 2024-05-09:
I believe the correct settings are on Roman's GitHub. Marc Hoffmann provided a diff for that model, which works well.

Yury-MonZon, 2024-05-09:
> I believe the correct settings are on Roman's GitHub. Marc Hoffmann provided a diff for that model, which works well.

I meant for the noob wing.

MrD-RC, 2024-05-09:
So do I.

closed 2024-05-09 by Yury-MonZon