RomanLut/INAV-X-Plane-HITL #26: Any way to get debug data?

Yury-MonZon, 2024-05-17:
Is there any way that I can get debug data(log, printf or any other way) out of HITL so I can graph it in real time?
Thanks!

RomanLut, 2024-05-17:
No such functionality is available, you have to implement it yourself.

closed 2024-05-17 by RomanLut

RomanLut, 2024-05-18:
Ok, I might not understand your original question correctly and debug[8] is what you actually need.

There is an array of 8 debug variables in inav int32_t debug[8] https://github.com/iNavFlight/inav/blob/ee747d66715b4bc9702c49c8b4c4b122cca4f68d/src/main/build/debug.h#L24

They can be written to blackbox, visualized in inav configurator in Sensors tab, or drawn on graph in HITL.

They can be used to debug various systems by setting **debug_modes** in cli or to output debug values in custom inav build.

Just assign debug[0] in inav custom built and value will be shown in HITL https://github.com/RomanLut/INAV-X-Plane-HITL/blob/main/doc/development.md#plugin-data-ref-editor-show-datarefs